SymbiFlow/sphinxcontrib-hdl-diagrams #10: Problems installing the plugin

oharboe, 2020-04-15:
I'm having problems installing the plugin:

```
$ pip3 install --user sphinxcontrib-verilog-diagrams
ERROR: Could not find a version that satisfies the requirement sphinxcontrib-verilog-diagrams (from versions: none)
ERROR: No matching distribution found for sphinxcontrib-verilog-diagrams
```

mithro, 2020-04-15:
The package is not yet published on pip, it is on my todo list.

For now you will need to do `pip3 install --user git+https://github.com/SymbiFlow/sphinxcontrib-verilog-diagrams.git#egg=sphinxcontrib-verilog-diagrams`

However, as this tool requires `yosys`, `netlistsvg` and other tooling not provided through pip, the recommended way is to actually install using conda. An example of doing this as part of Sphinx can be found in the https://github.com/SymbiFlow/python-symbiflow-v2x/tree/master/docs repository.

Would be good for us to update the README with this information!

mithro, 2020-06-23:
Closing as this is effectively a duplicate of #7 .

closed 2020-06-23 by mithro

mithro, 2020-06-24:
Package should now be up on PyPi!