SymbiFlow/sphinxcontrib-hdl-diagrams #14: Usage of -compat-int on yosys should depends on its version

daniellimws, 2020-04-18:
#13 did not account for the fact that older versions of yosys do not have the `-compat-int` option for `write_json`. This causes this extension to fail when an older version of yosys is used (in particular the version of yosys currently present on conda does not have this option).

The version of yosys should be checked to determine whether the `-compat_int` options should be added to `write_json`. 

https://github.com/YosysHQ/yosys/commit/20ce4118da1a0b14e8307aa0a605101252028757 is the commit that introduced this option.

mithro, 2020-04-18:
@daniellimws - I would say we just drop support for any version of Yosys that is too old? IE The minimum yosys version required is XXX where that is after the `-compat_int` was added?

daniellimws, 2020-04-19:
Ok. There should be no problem with this once yosys in conda is updated.

closed 2020-04-19 by daniellimws