SymbiFlow/sphinxcontrib-hdl-diagrams #22: Use material theme for readthedocs

daniellimws, 2020-04-26:
Currently the documentation is not using the material theme like other SymbiFlow projects.

I've changed my local setup to use the material theme for readthedocs, but the code block with line numbers looks awkward.

[image]

@mithro What do you think?

mithro, 2020-04-26:
@daniellimws - Actually that might be why the material theme was disabled....

daniellimws, 2020-04-26:
I'll change it to SymbiFlow's fork and fix the styling.

mithro, 2020-04-27:
I created an issue on the actual theme we are using at https://github.com/SymbiFlow/sphinx_materialdesign_theme/issues/2

PR #23 (2020-06-23): Convert readthedocs to use material theme
Close #22 

Currently the code blocks with the line numbers look awkward, the line numbers column is so wide. Should we keep them?

[image]
closed 2020-06-23 by mithro