SymbiFlow/sphinxcontrib-hdl-diagrams #40: Investigate if we can use the Yosys uploaded to PyPi

mithro, 2020-06-24:
@whitequark has been looking at how to get Yosys onto PyPi in a nice way to enable https://github.com/nmigen/nmigen

See if we can also [optional] use that package for diagram generation rather than the system or conda provided Yosys.

mithro, 2020-06-24:
Further information
 - https://yowasp.github.io/ && https://twitter.com/whitequark/status/1275311099475353601?s=20
 - Also https://twitter.com/whitequark/status/1275460960585764864?s=20

mithro, 2020-06-24:
@whitequark -- Any advice? We want to use Yosys to generate svg diagrams directly and json for netlistsvg from verilog files.

whitequark, 2020-06-24:
Sure, I think that should work. Are there any issues you've encountered?

mithro, 2020-06-24:
Haven't tried it yet. Hopefully @daniellimws will have some time to look doing it.

daniellimws, 2020-07-01:
Thanks to https://github.com/YoWASP/yosys/commit/aae50ce6a1e40861e2d576a2676013ae22ea9b9c this should be straightforward. @whitequark when will the next release to PyPi be? Currently this is not in the release build yet right?

whitequark, 2020-07-01:
As soon as https://github.com/YoWASP/yosys/actions/runs/153827693 finishes

daniellimws, 2020-07-01:
Wow that's fast. Thanks!

whitequark, 2020-07-01:
Should be released now. Make sure, if you add a dependency to the package, to use something like `>=0.9.post4547.dev9`.

closed 2020-09-14 by rw1nkler