SymbiFlow/sphinxcontrib-hdl-diagrams #43: Readme Rst is broken

mithro, 2020-06-25:
https://travis-ci.com/github/SymbiFlow/sphinxcontrib-verilog-diagrams/builds/172946780

```
Checking dist/sphinxcontrib_verilog_diagrams-0.0.post79-py3-none-any.whl: FAILED
  `long_description` has syntax errors in markup and would not be rendered on PyPI.
    line 81: Warning: Bullet list ends without a blank line; unexpected unindent.
Checking dist/sphinxcontrib-verilog-diagrams-0.0.post79.tar.gz: FAILED
  `long_description` has syntax errors in markup and would not be rendered on PyPI.
    line 81: Warning: Bullet list ends without a blank line; unexpected unindent.
```

Guess we should figure out how to test for that in a pull request?

daniellimws, 2020-06-25:
https://packaging.python.org/guides/making-a-pypi-friendly-readme/#validating-restructuredtext-markup

daniellimws, 2020-06-25:
Oh that's what travis is currently doing

PR #44 (2020-06-25): Add twine check to make build and fix rst error
Fix #43
closed 2020-06-25 by mithro