SymbiFlow/sphinxcontrib-hdl-diagrams #48: Rename to sphinxcontrib-hdl-diagrams and allow other inputs?

whitequark, 2020-06-27:
For example RTLIL or nMigen.

mithro, 2020-06-27:
Send a pull request with an non-verilog example and we will certainly do the rename!

whitequark, 2020-06-27:
Sorry, I don't have time to work on that in close future—I'm really busy with nMigen. Do you think you can find someone else to work on this?

mithro, 2020-06-28:
@whitequark - I'd be excited to have it included but totally understand you might not have the time to do it yourself.

@daniellimws might be interested in giving it a go? 

I think the primary question will be how to get the nMigen input into Yosys and then call the `write_json` (or other similar commands)? We use the following for the verilog;
https://github.com/SymbiFlow/sphinxcontrib-verilog-diagrams/blob/8c16b8ffab789e6928cdfb34873421f50be1d24d/sphinxcontrib_verilog_diagrams/__init__.py#L228-L233

whitequark, 2020-06-28:
> I think the primary question will be how to get the nMigen input into Yosys and then call the `write_json` (or other similar commands)? We use the following for the verilog;

Since nMigen is already Python, it can easily provide you with RTLIL input (it's just a call to `nmigen.back.rtlil.convert(...)`). Then you could stuff that into your `run_yosys` function, it will detect the `.il` extension and load it as RTLIL.

closed 2020-09-16 by mithro